MangoAutomation/ma-core-public #328: Modbus did not resume reads after an communication outage

resotek, 2015-01-16:
During a LAN power outage, all Modbus values stopped updating on the graphical view page, as expected. However, once the outage ended, the Modbus device values did not resume updating. The value and the timestamp remained frozen for each datapoint, as displayed with the SimplePoint graphical view object. The expected yellow triangle warning of no updates was not displayed. The BACnet devices updated normally, even though they were affected by the same outage.

Communication resumed the moment a respective Modbus data source was disabled/enabled or when I/O logging was enabled for the datasource. This occurred in ma-core 2.5, but has also been observed in earlier versions.

Joelhaggar, 2015-01-16:
I'm almost 100% positive this has been resolve and you don't have the latest modbus module.  http://store.infiniteautomation.com/module/modbus  I'm sorry about that. There was an update recently and I didn't realize it didn't make it into the main download on the server.

resotek, 2015-01-16:
I have a metadatasource that reads Modbus points and uses point links to write other Modbus points. It stopped working, but the time stamps were up to date in the graphical view. After I restarted that metadata source, the normal behavior resumed. This is similar to this Modbus issue in that it occurred at the same time and the fix was also to restart the datasource. I don't know whether these issues are related.

Joelhaggar, 2015-01-16:
You also didn't get the updated Meta Data Source module.  http://store.infiniteautomation.com/module/meta  There was a mixup with our website and the latest files where not there.  I have resolved it but you will need to manually upgrade the Modbus and Meta Modules.

closed 2015-01-16 by Joelhaggar