MangoAutomation/ma-core-public #330: Modbus - Delay between polls

andreisalomao, 2015-01-27:
Some modbus devices does not respond well when multiple requests are done with no delay between them. An option for modbus data sources could be added to configure the Delay Between Polls (from zero, meaning no delay, to something like 1000 ms).

resotek, 2015-02-20:
Similarly, it would be useful to be able to define the polling interval based on the time the reply is received rather than just based on the time the poll was requested. The user would choose either "Interval polling" or "Delayed polling."

This would allow polling Modbus devices based on their response time rather than on the Modbus Master's defined interval. For instance, if the Modbus device responds within 100ms and the "Delay Between Polls" is set to 50ms, then polling would take place every 150ms. If a Modbus response is received sooner, then the next poll goes out sooner. This is useful for attaining the maximum polling rate while avoiding error such as "Poll aborted because previous poll did not complete." The quantize option may not be possible when delayed polling is chosen over interval polling. The timeout option should still work.

Puckfist, 2018-01-23:
This is possible from a scripting data source in the latest Mango. Simply set the data source's poll to be VERY long, then you could have a script like,

while( true ) {
  if( RuntimeManager.refreshDataSource("DS_XID") != 1 )
    LOG.error("Failed to refresh data source, consider exiting poll loop.");
  RuntimeManager.sleep(5000);
  if( exitConditionPoint.value )
    break;
}

I am not sure there is a compulsion to implement this more fully.

Puckfist, 2018-02-05:
Closing as using a script should be sufficient.

closed 2018-02-05 by Puckfist