MangoAutomation/ma-core-public #337: Long Running Requests block data sources

terrypacker, 2015-02-19:
Long API requests block data sources - WARN  2015-02-03 16:48:13,880 (com.serotonin.m2m2.rt.dataSource.PollingDataSource.scheduleTimeout:106) - Dashboard Demo: poll at 2015/02/03 16:48:12 aborted because a previous poll started at 2015/02/03 16:48:12 is still running

terrypacker, 2015-02-19:
This may be fixable by moving spring to use Asyc type requests through Jetty instead of the current synchronous methodology.

mlohbihler, 2015-02-22:
This behaviour is by design. It was deemed undesirable to have a device polled multiple times simultaneously, especially since some data sources will block requests behind each other causing outstanding requests to pile up. The message is intended as an indication that the polling rate on the data source is too high.

jazdw, 2015-02-22:
The issue is that long running REST API calls are blocking the data sources somehow. The API calls are unrelated to the data sources, we are just retrieving point values from the database. It may be that the contention is occurring at the database layer.

terrypacker, 2016-05-26:
This should be checked on a graphical view page that access a high polling virtual data point logging into the NoSQL database.

terrypacker, 2016-06-03:
Closing as this doesn't seem to be happening anymore in any of my tests.

closed 2016-06-03 by terrypacker