MangoAutomation/ma-core-public #338: Modbus Log I/O option does not log outgoing Modbus traffic when the target device is unresponsive

resotek, 2015-02-20:
The Modbus Log I/O option should log outgoing Modbus traffic as soon as it is sent, but apparently it logs nothing at all when the target device does not respond to the Modbus poll. An easy test case is to v change the datasource to an invalid IP address as you tail the log file, then change it back to the valid IP address.

terrypacker, 2015-03-18:
From your example test there is nothing to log.  The connection is failing on the Open Socket request which is a TCP/IP layer request and has nothing to do with the Modbus Protocol Spec, basically we don't have anything we can log.

All Outgoing messages appear to be logged when they are sent.  Perhaps I am missing something.  Can you provide another example of when you do not see the O:xxxxxxx messages in the log?

resotek, 2015-03-18:
Can you log each time the Open Socket request is attempted, or at least each time it fails? I would expect that, for instance, with a 5-second polling interval I should see something appended to the Modbus log file every 5 seconds, even if I unplug the Modbus device from the LAN.

Puckfist, 2017-04-11:
This information is available by enabling debug level in the com.serotonin.modbus4j package, the IOLog will remain only characters that have actually been transmitted, closing this issue.

closed 2017-04-11 by Puckfist