MangoAutomation/ma-core-public #346: Bash scripts have multiple area's that can be improved.

JosephMillsAtWork, 2015-03-08:
All of the bash scripts have .sh at the end of them.  This is not needed, 
The shebangs are not using env and should for cross platform reasons, 
Most of the test's that are happening n strings are not enclosed in double square. 
more explanations and better headings. along with help sections 

references: 
http://mywiki.wooledge.org/BashGuide/CommandsAndArguments#Scripts
http://mywiki.wooledge.org/BashPitfalls

jazdw, 2015-03-08:
I think its best to leave the shell scripts with a .sh extension. This helps identify them at a quick glance especially on Windows and also serves as a way to force their line endings in .gitattributes files.

The scripts probably should just use /bin/sh rather than hardcoding bash. You should check they are compatible with shells such as dash which is the default shell for Debian & Ubuntu. I'm not sure using /usr/bin/env sh is more appropriate than just /bin/sh as practically everything I'm aware of should have /bin/sh and who's to say /usr/bin/env exists?

JosephMillsAtWork, 2015-03-09:
*.sh scripts are for sh scripts simple as that.  If you would like we could name them like "whatever.bash"
But I would rather get rid of the extension because it  not really a sh script.

The shebangs have been fixed to make sure if bash is not installed to /bin/bash it will still run.

The Strings that where getting tested against are now all double square so there is not false errors, 
 
Help section is there and so is the better headings.

jazdw, 2015-03-09:
/bin/sh is not a specific shell, its linked to whatever the default shell for the system is. i.e. some systems may have /bin/sh linked to bash, some to dash. A script using a /bin/sh shebang should be generic and compatible with all shells (within reason). Its pretty standard to just use .sh for these generic type scripts.

I'm proposing to test the scripts against dash (which is the default for MangoES as it uses Debian) and remove any bash specific stuff.

JosephMillsAtWork, 2015-03-09:
Jazdw you know that I am the author of mangoES operating systems correct ?  Like I wrote them.and maintain them,

jazdw, 2015-03-09:
Hi Joseph, I believe the MangoES uses the Debian operating system if I am not mistaken. But no, I was not aware that you were responsible for setting up the MangoES. Good to know but I'm not sure why that matters in the context of this conversation, my comments were not intended as a personal attack. I'm just saying if the intention of these changes are to make the scripts more cross-compatible we should make the scripts shell generic.

JosephMillsAtWork, 2015-03-09:
Hey Jazdw, 
I was not taking it as a personal attack at all by any means just wanted to clarify that I know a little bit about the OS and how it is put together.  :)  My intentions with the scripts is to 
     1) make sure that they work (starting stopping restarting ect)
     2) clean it up some along with adding in some features. (help verbose ect)
     3) test test test :) 
     4) merge 

thanks for all the info. cya around :)

jazdw, 2015-03-09:
Righto, no worries.

I'm Jared by the way, I work for Deltamation in Australia. As a bit of background we have been using Mango for a couple of years and Joel has been getting me to do a bit of work recently on some dashboard modules, the REST API, and a configuration module for setting IP addresses etc on the MangoES.

terrypacker, 2015-08-19:
Closing as complete enough for 2.6.0

closed 2015-08-19 by terrypacker