MangoAutomation/ma-core-public #347: Bash Scripts PID is not getting killed on stop

JosephMillsAtWork, 2015-03-08:
because the bash scripts are reading the pid for a text file it does not always kill mango in the most graceful ways.  Also there is no checker to make sure that it is killed afterwords

jazdw, 2015-03-09:
What's the issue with reading the PID from a text file?

I'm not sure these shell scripts are the best idea for starting and stopping Mango TBH. We have been using YAJSW (http://yajsw.sourceforge.net/) with a lot of success. Its quite nice in that you supply a configuration file and it will build and install a System V init script for Linux or install Mango as a service on Windows. It basically starts up a little Java app that monitors the main process and can restart it if it crashes etc.

Joelhaggar, 2015-03-09:
Hey Jared,  to fill you in on a few things that you can also think about.

We have been having a lot of problems with the current start up scripts which is why Joseph is looking at them.  Some time when you click the restart button in Mango it doesn't restart.  Then we also have the problem of being able to stop mango.  Sometime when you do ./ma.sh stop it just says it doesn't find a PID file but Mango is still running and it can be a real pain to get Mango to actaully stop.

I have used YAJSW on windows and it works great but never tried it on Linux.  Unless we automate it, it's obviously way too much extra work to expect a new user to get into but maybe there is a way to make that easier.  The other issue with using YAJSW is that the new auto upgrade stuff won't work because the upgrade script isn't going to get called so we would need to solve that which might be able to be done by adding the upgrade stuff to Mangos startup.

JosephMillsAtWork, 2015-03-09:
Framework is up with new push.,

JosephMillsAtWork, 2015-03-09:
Hey Jared,
Hey Joel,
Nice too meet you.  I have already made many mango packages(deb) and one of them is a init script and also a systemd script.  but these are still calling the BASH shell files.

jazdw, 2015-03-09:
Yeah I have definitely noticed the issues with ma.sh start and stop. There's not really an issue with the methodology of getting the PID from a file though, its more likely a bug in the script. Storing a daemon PID in a file is a tried and tested pattern.

We could include a YAJSW configuration file which could be used to install the service on Windows and also a some pre-built init scripts for Linux so the end-user doesn't have to do generate their own. I'm not sure how the new auto upgrade scripts work but I'm sure they could be integrated somewhere. Just something to think about anyway, rather than duplicating functionality which is available elsewhere.

PS I've used YAJSW on Linux and it works nicely, I think it was necessary to set wrapper.fork_hack = true to avoid issues (http://yajsw.sourceforge.net/YAJSW%20Configuration%20Parameters.html)

JosephMillsAtWork, 2015-03-09:
Jazdw there is a debian and ubuntu repository that has all this

https://github.com/infiniteautomation/IASDevelopment/blob/master/mangoDebianPackages/mango-2.5.2.2/debian/mango/etc/init.d/mango 

There is some other packages in there have a look :)

jazdw, 2015-03-09:
OK cool I'll have a poke around in there at some point. Just real quick, why not just link ma.sh into the init.d directory?

JosephMillsAtWork, 2015-03-09:
jazdw,
Hey there there are many reasons that I do not want to do that.  manly for simplicity and I did not know at the time of writing the init script if ma.sh or any of the other scripts where also checking to make sure that they are links -L thus that could have given me back something not so swell.  I also did want to get around to hacking on the bash files in the core.  But I have been busy with other things as you can see if you look at the commits in that private branch. 

Other reasons.  the control during boot, esp with some of the eshelper stuff that I have been working on.  Like some of the touch stuff and what not.  in other words I wanted to order the way that it i was getting processed by.

I also was putting this on a lot of ES machines so I wrote ssh scripts for that.

The Repo is not that great at the moment and there are bugs in a lot of the packages (spelling error , reconfigure errors ect).  But the whole 
HIERARCHY 
Is starting to come together wither that be burning and licencing 100's of ES machines in a day and keeping track of that,  or well you can see the packages that are in there,  Or configuration management system's ect.  

Super stoked about all that. But would like to start to get my hands more dirty in mango-core and what not. 

cheers,

Joseph Mills

terrypacker, 2015-08-19:
Closing as complete enough for 2.6.0

closed 2015-08-19 by terrypacker