MangoAutomation/ma-core-public #351: Too Many Alarms can be generated from Poll Aborted Alarm

Joelhaggar, 2015-03-11:
If a data source starts aborting polls an alarm is raised each time which can generate 100's to 1000's of urgent alarms and emails.

I'm not sure the best way to handle this but maybe some sort of delay on the same event being triggered over and over.

jazdw, 2015-03-11:
Yeah I've noticed this as well, I tried to fix it before but it obviously didn't work - cbdc68320bf879f6e488c0b361d685f135de324e

terrypacker, 2015-03-11:
I will take a look at this today.  Jared, is all you did change the Duplicate Handling on the Event Type in the DataSourceVO class?

jazdw, 2015-03-11:
Nah there were changes in PollingDataSource as well as per the commit I linked to

terrypacker, 2015-03-11:
Oh sorry I didn't see that link to the commit.  Thanks.

terrypacker, 2015-03-12:
Ok so what is happening is that the Event is being fired on every first Missed poll but it is using the total number of missed polls in the alarm.

So the first alarm will say it has aborted 1 poll, but the counter will keep incrementing until a successful poll is reached.  At that time the alarm goes inactive.

The next time it misses a poll it will say it missed (number of polls aborted last time) + 1.

The problem you are seeing is that the alarm resets after the first successful poll.  So on data sources where there are 1 good 1 aborted in a row you will see one Event every time.  The alarm does not generate duplicate events.  Each event happens after it has a successful poll.

This alarm was put there because of scenarios like this that I was seeing in a lot of installations where people set the poll period too fast to do the work required.  It alerts them to a larger problem.

terrypacker, 2015-03-12:
Also note that Jared's fix will only be in Mango Version 2.5.2 as it was committed after the 2.5.0 release.

terrypacker, 2015-09-02:
Closed as completed.

closed 2015-09-02 by terrypacker