MangoAutomation/ma-core-public #352: No Change and No Update Event Detectors - False Alarms?

terrypacker, 2015-03-11:
The "no change" and "no update" detectors seem to be triggering false alarms: today I was creating several points on the OPS Modbus datasource that has a 1000ms polling interval, and setting the logging type to "All data." I received an "unchanged" alarm and a "no update" alarm regarding XID=DP_VPC1-1s. 

The alarm finally cleared after I disabled then re-enabled the datapoint. As far as I remember, I had not disabled or updated the datasource itself today, but I had saved changes to the point plotted above. Before I disabled the datapoint at 17:46:49 I saw that there were 10 medium priority work items, but according to the DP_medpri history, it had dipped to zero several times while the alarm was active. I believe this indicate that the alarm was not caused by excessive or un-executed medpri work items.

Puckfist, 2015-12-13:
They do indeed. With a Virtual Data Source polling at 750ms and a point set with no update and no change detectors set to 3 seconds, they will fire every time you pretty the 'save point' button.

Puckfist, 2016-01-08:
My suspicion is that these were triggered by loading the last logged value, while computing the state of these detectors happens in the runtime. In that case does it make sense to have these event detectors test their initial state on wakeup? If not we could override DifferenceDetectorRT's initializeState in both the NoUpdate and NoChange detectors to not check at initialization time.

I suspect I overlooked interval logging. In making my previous comment

Puckfist, 2016-01-09:
Okay, I looked into it, and surely they can cause false alarms. The trick would be to occupy the system sufficiently that the medium pool isn't executing (see DataPointRT.EventNotifyWorkItem: 586) because those two detectors lazily disregard the time from the pointValue (see NoUpdateDetectorRT and NoChangeDetectorRT:21) where their super uses the current system time to determine when the changes occur to the point. I bet raising the event occurs in the high pool.

Also, the NoChangeDetectorRT seems totally unimplemented. This whole setup is pretty elegant (challenges for computational resources and ignoring the newValue.getTime() to possibly get closer closing this issue not withstanding), but does the NoChange need an if(! oldValue.toString().equals(newValue().toString()) ) to deliver on its post condition?

Puckfist, 2016-01-09:
Nope, more looking. DataPointRT makes that call and only if !PointValueTime.equalValues(oldValue, newValue)

Well, the thread pool explanation could still certainly be an issue.

resotek, 2016-01-20:
I certainly receive NoChange events often. This certainly happens when there are medium priority work items. Often the events last a few seconds.

Puckfist, 2016-01-30:
Another observation I had while looking at this stuff is that we could use these to determine a certain amount of system load. If we pass in the times for the point values and these differ significantly from the system time, we could conclude that the medium pool is suffering.

Puckfist, 2016-07-05:
There is also an issue that the No Update/No Change detectors give false alarms when the data source has been disabled for more than the no update/change period at the moment the datasource is enabled.

Puckfist, 2017-03-27:
Closing as the no update no change situation where the data source or Mango is off is actually valid, those data points did not receive updates or changes during that interval

closed 2017-03-27 by Puckfist