MangoAutomation/ma-core-public #353: Data Source Point Updates while Running - Changes optionally picked up

terrypacker, 2015-03-12:
When a data source's points are updated while it is running there is no guarantee that the changes will make it into the running data source.

All data sources should implement a way to check the addedChangedPoints List and update points accordingly.

Test and review the Modbus DS first, as it appears this is not being done there.

terrypacker, 2015-03-12:
This does not seem to be the case for any Polling Data Source, Modbus is a polling data source so it inherently implements the Change pickup logic.

Puckfist, 2015-12-15:
Weird... I just experienced a situation on a system where the permissions had been set through the bulk update method, and the modbus data source's points (gotten through runtimeManager by id) was seemingly not informed of a permissions change early the next morning. I disabled and re-enabled the data source to get the changes observed. Is the API bypassing the runtime manager there, then?

terrypacker, 2015-12-15:
Yes it is bypassing the runtime manager.  We will need to modify it to know what points are updated and then force a restart of those points.

terrypacker, 2015-12-18:
Closing this as not a problem and opening new issue for Permissions not being picked up.

Puckfist, 2016-02-08:
Reopening. I experienced with the serial data source that when I saved the modifications to a point that had already been running that it's changes were not picked up. I had to save the datasource to get the modifications to the point.

terrypacker, 2016-02-09:
Closing this and opening new issue #636 as this issue is specific to the Serial Data Source.

closed 2016-02-09 by terrypacker