MangoAutomation/ma-core-public #359: add persist path to sstGraphics for the custom folder

Joelhaggar, 2015-03-23:
Add a folder under web/graphics called custom.  Make this folder persist on a upgrade so that users can upload their own image sets without loosing them.

Alternatively we could make a new module called CustomGraphics that is empty and the whole graphic folder on the persist path.

terrypacker, 2015-04-10:
To do this we either need to change the way the graphics property works for the module.properties file so that it allows more than 2 directory or as you said create a new module that persists its graphics directory.

terrypacker, 2015-10-15:
This can be changed by modifying the ViewGraphicLoader, ImageSet, DynamicImage classes.  Here is a start on the ViewGraphicLoader:

/*
    Copyright (C) 2014 Infinite Automation Systems Inc. All rights reserved.
    @author Matthew Lohbihler
 */
package com.serotonin.m2m2.view;

import java.awt.Container;
import java.awt.Image;
import java.awt.MediaTracker;
import java.awt.Toolkit;
import java.io.File;
import java.io.FileInputStream;
import java.net.MalformedURLException;
import java.util.ArrayList;
import java.util.Arrays;
import java.util.List;
import java.util.Properties;

import org.apache.commons.logging.Log;
import org.apache.commons.logging.LogFactory;
import org.eclipse.jetty.util.resource.Resource;

import com.serotonin.m2m2.Common;
import com.serotonin.m2m2.Constants;
import com.serotonin.m2m2.module.Module;
import com.serotonin.m2m2.module.ModuleRegistry;
import com.serotonin.m2m2.web.OverridingFileResource;

public class ViewGraphicLoader {
    private static final Log LOG = LogFactory.getLog(ViewGraphicLoader.class);

    //    private static final String GRAPHICS_PATH = "graphics";
    private static final String INFO_FILE_NAME = "info.txt";

    private static final String IGNORE_THUMBS = "Thumbs.db";

    private final String path = Common.MA_HOME + File.separator + Constants.DIR_WEB;
    private final String overrides = Common.MA_HOME + File.separator + "overrides" + File.separator + Constants.DIR_WEB;
    private List<ViewGraphic> viewGraphics;

    public List<ViewGraphic> loadViewGraphics() {
        viewGraphics = new ArrayList<ViewGraphic>();

        for (Module module : ModuleRegistry.getModules()) {
            if (module.getGraphicsDir() != null)
                loadModuleGraphics(module);
        }

        return viewGraphics;
    }

    private void loadModuleGraphics(Module module) {
    	OverridingFileResource resourcePath = null;
		try {
			Resource overrideResource = Resource.newResource(overrides + module.getWebPath() + File.separator + module.getGraphicsDir());
	    	Resource moduleResource = Resource.newResource(path + module.getWebPath() + File.separator + module.getGraphicsDir());
	    	resourcePath = new OverridingFileResource(overrideResource, moduleResource);
	    	
	    	String[] dirPaths = resourcePath.list();
	        for (String dirPath : dirPaths) {
	            try {
	            	Resource dir = resourcePath.addPath(dirPath);
	                if (dir.isDirectory() && !dir.getName().startsWith("."))
	                    loadDirectory(dir, module.getName() + ".");
	            }
	            catch (Exception e) {
	                LOG.warn("Failed to load image set at " + dirPath, e);
	            }
[... 77 lines cut ...]
                throw new Exception("Invalid type: " + typeStr);

            viewGraphics.add(g);
        }
    }

    private String getProperty(Properties props, String key, String defaultValue) {
        String prop = (String) props.get(key);
        if (prop == null)
            return defaultValue;
        return prop;
    }

    private int getIntProperty(Properties props, String key, int defaultValue) {
        String prop = (String) props.get(key);
        if (prop == null)
            return defaultValue;
        try {
            return Integer.parseInt(prop);
        }
        catch (NumberFormatException e) {
            return defaultValue;
        }
    }
}

terrypacker, 2016-10-28:
Done and done, added ability to have multiple graphics directories. Ready for testing.

Puckfist, 2016-11-07:
I have confirmed that this is working, but I discovered in the courses of testing that you have to restart Mango to pick up changes to the images in this directory? Probably not 2.7.x important, but I'll open an issue if there isn't one already.

closed 2016-11-07 by Puckfist