MangoAutomation/ma-core-public #364: Simulation Data Source

terrypacker, 2015-03-27:
We could create a new type of Simulation Data Source.  This data source type would be able to read logged point values out of the database and pretend they were the values being read by it.  We could then simulate a Modbus data source starting from some time.  The simulation would then  use the starting time (in the past) and start pulling out the data one by one using the time difference between samples.  For example the simulator would read the start value and then save it into its point stream, then it would read the next point value and determine the time difference between it and the current value.  Then it would set a timer and wait that long until pulling in the next value.  This would allow for a variable time input that is essential since some point values can be inserted by a user in the UI or other ways.

We could expand on this by streaming Events/Alarms and actioning the Event Handlers etc.

This could be a very powerful debugging tool to re-create various situations that currently we can only review in logs and event tables.

Puckfist, 2017-10-02:
The script data source can probably do what this is about. Currently one could do something like this...

var data = p.pointValuesSince(0);
var lastTime = data[0].time;
for(var k = 0; k < data.length; k+=1) {
  RuntimeManager.sleep(data[k].time - lastTime);
  target.set(data[k].value);
}

or if one was feeling crazy, one could invoke the query with a callback more directly (after all, Java right from Nashorn). I think we can close this, but what say ye?

terrypacker, 2017-10-10:
I'm ok with closing this.

closed 2017-10-10 by terrypacker