MangoAutomation/ma-core-public #371: Incremental NoSQL Backup - Versions to keep is deleting files even when using Incremental option

Joelhaggar, 2015-04-07:


terrypacker, 2015-04-08:
I was aware this would happen and left it in on purpose.  I think just because you are doing incremental backups doesn't mean you want to keep all of your historical data.  I think it should be left in.  We could add an additional setting that disables this feature when checked.

Joelhaggar, 2015-04-08:
I think you are right it makes sense to limit the number of backups to keep but it should probably be really high by default like 365 or 730 or something.  Main thing is that the user understands it so they don't think they have backups but in-fact they are being deleted.  Maybe rename it or put a warning sign by it with a popup explanation

terrypacker, 2015-04-10:
Added red text comment when using incremental backups.
Also set the default number of backups to keep to 365.

Waiting to close this issue until we review this again.

closed 2015-09-09 by terrypacker