MangoAutomation/ma-core-public #394: ProcessLog - performance degradation from synchronization

terrypacker, 2015-05-22:
When running multiple process loggers there appears to be some locking problems with the write method in the ProcessLog class.  This was seen on a very large system with 1 Persistent publisher logging and 4 Persistent data source's logging.

Puckfist, 2018-03-19:
Huh. It looks like it's just a synchronized block on a member variable, and every data source or publisher should have created their own process log, so they shouldn't be blocking each other. That synchronization is so that the various data sync tasks can share the same process log, which isn't an issue?

I guess maybe some kind of concurrent linked queue could be faster if the log load is heavy, but, that seems odd.

terrypacker, 2018-04-04:
Closing as irrelevant.

closed 2018-04-04 by terrypacker