MangoAutomation/ma-core-public #395: Nightly Purge - SQL Point Values

terrypacker, 2015-05-28:
At a large site the following query was taking several hours to complete:

DELETE FROM pointValues WHERE dataPointId NOT IN (SELECT ID FROM dataPoints) LIMIT 1000;

Investigate additional indexes or re-working the query.

minghu, 2015-05-31:
I suggest @deprecated this funcion,this delete actually take long time,And we can from  use case design to prevent this condition from happening,instead of when it happen then to delete it(if it is easy,that is ok)

terrypacker, 2015-07-30:
I Agree, this can be removed and we can delete orphaned values when they happen on any point that is deleted.  We can create a background task that is run after a point is deleted.  This query will not use the IN so it should be much quicker.

terrypacker, 2015-09-02:
Closing as fixed, as of core 2.6.0 there is a boolean system setting: "enablePurgePointValues" that will disable the purging of values.

terrypacker, 2015-09-02:
The setting is available under the Purge settings on the system settings page.

closed 2015-09-02 by terrypacker