MangoAutomation/ma-core-public #403: NoSQL Module Shutdown Timer - Cancel if license is upgraded

terrypacker, 2015-06-25:
Currently the NoSQL module will still shutdown Mango when license is added during runtime.  We need a hook to notify any modules that a license has been added.

- A fix would inform both the TimedExpiryChecker and the UsageExpiryChecker of this condition as to check the license and disable the timer or usage counter.

terrypacker, 2015-08-05:
Fixed by adding a new License Definition to core and using it in the NoSQL,Brewers Dashboard and Excel.

closed 2015-08-05 by terrypacker

Joelhaggar, 2015-08-06:
I forgot to follow up on this. I think the excel reports just shouldn't allow more reports than are licensed and we should remove the demo reports. I'm not sure if it works this way but I don't think it's a good idea to shut down as then any user with access could cause mango to shut down. 

Joel.
Infinite Automation Systems Inc.
(303) 565-6641