MangoAutomation/ma-core-public #409: 2.6.0 REST - Use of dot/period in URLS needs review

terrypacker, 2015-07-25:
Spring is set to automatically remove the extension from the URLs, this is causing problems for usernames with dots in them such as bob.yourhead 

1.  This only effects endpoints where we capture the path parameter as the last element on the path:
/rest/users/{capture}

This can be fixed by adding /rest/users/{username.+} to any endpoint we wish to capture a dot. We must then ensure all javascript calls to the endpoint do not contain a suffix.  So where we would usually do this:

GET
/rest/users/bob.yourhead.json

We must now do this:
GET
/rest/users/bob.yourhead

The other option is to turn off the suffix truncation feature of Spring so that we receive the entire string in the request.

jazdw, 2015-07-27:
OK as a follow up to my email, I think we should turn the suffix truncation off. We will just have to check that it doesn't break anything else. Reference for how to turn it off - https://jira.spring.io/browse/SPR-5778

I tried your /rest/users/{username.+} syntax and it didn't work. Also tried percent encoding the dot.

I've removed the .json suffix from all the JS API methods, I believe everything is still working correctly.
infiniteautomation/ma-modules-private@8c364f60fcba4078a45db68c031b8a42730b8361

terrypacker, 2015-07-27:
I've modified the MangoRestSpringConfiguration class to stop using suffix matching.  The main concern will be any file upload/download links as previously we could use the suffix to determine the file type.  Now we will have to ensure the content-type headers are properly set.

terrypacker, 2015-08-19:
Closing as everything seems to be working after futher testing.

closed 2015-08-19 by terrypacker