MangoAutomation/ma-core-public #417: WebSocket - Provide a hook for notification that Mango's web server is shutting down.

terrypacker, 2015-08-20:
This will allow the web sockets to terminate gracefully.  Currently they just thrown an exception after the web server is down but the event and runtime managers are still stopping.

Puckfist, 2015-12-10:
Was this an issue on the client side? I just ended the web server politely and -9 style, no errors to console and just a refused connection on the client side. Is this closed?

terrypacker, 2015-12-10:
This is more of a feature where it would be good to send a shutdown notification message out on all web sockets so that the frontend could notify the user that this is happening.

terrypacker, 2015-12-10:
Sorry this issue can be closed if there is no longer a problem.  We should open another for the above feature.

Puckfist, 2015-12-10:
Still seems like that feature is covered in this issue as titled