MangoAutomation/ma-core-public #425: Add the value of the data point into an email fired from an event detector.

Joelhaggar, 2015-09-09:


terrypacker, 2015-09-09:
This could be a little tricky, the data point detectors don't necessarily track the current value of a point, so it isn't easily retrieved at the place in the code where the event is raised.

One option is to re-work all the event detectors to track the current value, the other option is to retrieve the value for the data point when the email is created.  This may produce a slight amount of lag/difference between the value of the point when the alarm was created vs. then the email is sent.

jeremyjhall, 2015-09-10:
If I can chip in my 2c here, if you're going to get the event detector to retrieve the current value, it may be worth considering adding an optional 'include previous x count of point history' to be included in the email/alert along with the current value. This may significantly increase the usefulness of the alarms as the user can see the values before the detected condition was met.

terrypacker, 2015-09-11:
Thanks for the input, I'm adding that feature today.

terrypacker, 2015-09-21:
Done.

closed 2015-09-21 by terrypacker