MangoAutomation/ma-core-public #428: Modifications for CRON and Quantize for all Polling Data Sources

terrypacker, 2015-09-16:
Steps to Complete:

1.   Add String for cron pattern to all polling data source VOs and a boolean for Quantize if using time period polling.

2. Modify validation for VOs to confirm that if cron pattern is not null then it must be valid

3. Modify JSON export only print out either cron pattern or poll period settings

4. Modify JSON import to confirm that either a poll period or a cron pattern are present

5. Create a new JSP snippet that combines the poll period selection with Quantize boolean and cron pattern widget and shows hides each according to checking a box.  Then add this to the data source edit page.

6. Modify the Data Source RT to set either the poll period with optional Quantize or the Cron pattern.

Puckfist, 2015-12-12:
Attention should be given to [424](https://github.com/infiniteautomation/ma-core-public/issues/424) while at it.

terrypacker, 2016-04-12:
Consider making an abstract class PollingDataSourceVO that extends DataSourceVO.

Puckfist, 2016-07-06:
It looks like the scripting data source still needs this applied to it

Puckfist, 2016-07-06:
It also has a tendency to build cron expressions considered invalid. It's initial desire is to set the day of the month and the day of the week to * (Every?) but this is considered an illegal expression. The easiest way to see this is a new data source, click the 'Every 10' in the seconds column and then try to save the source.

Puckfist, 2016-07-06:
It looks like there may be many data sources that would still like this applied to them.

The 'set to poll period' button is greatly misleading or outright wrong. It works if and only if:
1) updatePeriods < timeUnit modulus (so, 60 for seconds/minutes, 24 for hours)
2) updatePeriods is a factor of timeUnit modulus

terrypacker, 2016-10-31:
The class PollingDataSourceVO and PollingDataSourceModel were truncated from the 2.9.x branch and should be retrieved from git history in when ready to implement this.

terrypacker, 2017-10-10:
Duplicate-ish of #699

closed 2019-04-09 by terrypacker