MangoAutomation/ma-core-public #430: Resolve cyclic maven dependency between core and core-private

jazdw, 2015-09-25:
There is a cylic maven dependency between com.infiniteautomation:mango and com.infiniteautomation.mango:ma-priv which should be fixed as it causes errors in Eclipse and causes general havoc triggering rebuilds etc.

com.infiniteautomation:mango does not actually depend on com.infiniteautomation.mango:ma-priv rather it is included as a dependency so that it is on the class path when running Mango and probably for the Ant build scripts too. Perhaps create another project which depends on mango and ma-priv for running and building mango using Ant.

terrypacker, 2016-05-25:
Pushing this off again as we still aren't doing release builds with Maven.

jazdw, 2016-05-26:
This can still be resolved without switching to Maven for builds, we just need another Maven project which depends on both com.infiniteautomation:mango and com.infiniteautomation.mango:ma-priv. Put your Core ant script in this project and tweak the ant script.

closed 2017-03-21 by Puckfist