MangoAutomation/ma-core-public #435: REST API login always logs out existing user

jazdw, 2015-10-13:
REST API login always logs out the existing user regardless of the logout header.

terrypacker, 2015-11-12:
I am looking at the code and testing it, this doesn't seem to be the case in 2.6.x, can you give me more information?

jazdw, 2015-11-12:
I was a using the latest core and module at the time, so it was 2.6. The way that it is supposed to function is that if a used is already logged in and the logout header is set to false then the existing user is not logged out. This is so that if you are logged in as admin for example and you have a dashboard setup to auto login as guest you aren't logged out of the admin account when you view the dashboard. If that's how its behaving then maybe it has been fixed.

terrypacker, 2015-11-12:
I re-wrote the login code recently by adding a LoginManager class so perhaps it was fixed during that.  I'll leave this open until you get some time to confirm that its working how you expect.  But in my test today if there was already a logged in user then it returns that user instead of logging in as the new user.

Puckfist, 2015-12-11:
I believe this is still an issue, but perhaps I am misunderstanding the expected behavior.
Currently, LOGOUT FALSE:
Computer 1 logs on as foo
Computer 2 uses the swagger login section to submit a login as foo, __logout false__
Computer 2 gets response code 200
Computer 1 refreshes and is still logged in.

LOGOUT TRUE:
Computer 1 logs on as foo
Computer 2 uses the swagger login section to submit a login as foo, __logout true__
Computer 2 gets response code 200
Computer 1 refreshes and is still logged in.

terrypacker, 2015-12-12:
Interesting interpretation, I was thinking this had to do with sessions on the same browser, so if I was currently logged in as admin and I went to a page with the code that has logout=true I would become the new user.  I think we need clarification on what this actually means.

Puckfist, 2015-12-12:
Ah, my bad. I tested that behavior and it's working as expected. I'm not sure what I was initially thinking is desirable behavior or not. I can also verify if Computer 1 and 2 are admin, and 2 does logout=true, 1 will remain admin and 2's other tabs will not. Marking as closed.

closed 2015-12-12 by Puckfist