MangoAutomation/ma-core-public #439: PointLink script broken in mango2.6 beta

jorgempy, 2015-10-20:
Hi, testing mango2.6 i found that the scripting in pointlinks fails.
i created a minimal testbed, it includes:
* create virtual dtp's a,b,c (numeric,settable)
* create two pointlinks 
     * a -> b no transformation
     * a -> c using example transformation "return source.value \ 2;"
go to watchlist, write some number to "a".
     * point "b" updates succesfully
     * point "c" throws **Could not convert result "sun.org.mozilla.javascript.InterpretedFunction@17c5dd53" to Numeric**

same thing works fine in mango2.5.
testbed JSON for importing: 
```javascript
{
   "dataSources":[
      {
         "xid":"DS_362292",
         "name":"bugtest",
         "enabled":true,
         "type":"VIRTUAL",
         "alarmLevels":{
            "POLL_ABORTED":"URGENT"
         },
         "purgeType":"YEARS",
         "updatePeriodType":"MINUTES",
         "updatePeriods":5,
         "editPermission":"",
         "purgeOverride":false,
         "purgePeriod":1
      }
   ],
   "dataPoints":[
      {
         "xid":"DP_829661",
         "name":"a",
         "enabled":true,
         "loggingType":"INTERVAL",
         "intervalLoggingPeriodType":"MINUTES",
         "intervalLoggingType":"AVERAGE",
         "purgeType":"YEARS",
         "pointLocator":{
            "dataType":"NUMERIC",
            "changeType":{
               "type":"NO_CHANGE",
               "startValue":"0"
            },
            "settable":true
         },
         "eventDetectors":[
         ],
         "plotType":"SPLINE",
         "unit":"",
         "templateXid":"Numeric_Default",
         "chartColour":"black",
         "chartRenderer":{
            "type":"IMAGE",
            "timePeriodType":"DAYS",
            "numberOfPeriods":1
         },
         "dataSourceXid":"DS_362292",
         "defaultCacheSize":1,
         "deviceName":"bugtest",
         "discardExtremeValues":false,
         "discardHighLimit":1.7976931348623157E308,
         "discardLowLimit":-1.7976931348623157E308,
         "intervalLoggingPeriod":1,
         "intervalLoggingSampleWindowSize":0,
         "overrideIntervalLoggingSamples":false,
         "purgeOverride":false,
         "purgePeriod":1,
         "readPermission":"",
         "setPermission":"",
         "textRenderer":{
            "type":"ANALOG",
            "useUnitAsSuffix":true,
            "unit":"",
            "renderedUnit":"",
            "format":"0.00"
         },
         "tolerance":0.0
      },
      {
         "xid":"DP_171102",
         "name":"b",
         "enabled":true,
         "loggingType":"INTERVAL",
         "intervalLoggingPeriodType":"MINUTES",
         "intervalLoggingType":"AVERAGE",
         "purgeType":"YEARS",
         "pointLocator":{
            "dataType":"NUMERIC",
            "changeType":{
               "type":"NO_CHANGE",
               "startValue":"0"
            },
            "settable":true
         },
         "eventDetectors":[
         ],
         "plotType":"SPLINE",
         "unit":"",
         "templateXid":"Numeric_Default",
         "chartColour":"black",
         "chartRenderer":{
            "type":"IMAGE",
            "timePeriodType":"DAYS",
            "numberOfPeriods":1
         },
         "dataSourceXid":"DS_362292",
         "defaultCacheSize":1,
         "deviceName":"bugtest",
         "discardExtremeValues":false,
         "discardHighLimit":1.7976931348623157E308,
         "discardLowLimit":-1.7976931348623157E308,
         "intervalLoggingPeriod":1,
         "intervalLoggingSampleWindowSize":0,
         "overrideIntervalLoggingSamples":false,
         "purgeOverride":false,
         "purgePeriod":1,
         "readPermission":"",
         "setPermission":"",
         "textRenderer":{
            "type":"ANALOG",
            "useUnitAsSuffix":true,
            "unit":"",
            "renderedUnit":"",
            "format":"0.00"
         },
         "tolerance":0.0
      },
      {
         "xid":"DP_432453",
         "name":"c",
         "enabled":true,
         "loggingType":"INTERVAL",
         "intervalLoggingPeriodType":"MINUTES",
         "intervalLoggingType":"AVERAGE",
         "purgeType":"YEARS",
         "pointLocator":{
            "dataType":"NUMERIC",
            "changeType":{
               "type":"NO_CHANGE",
               "startValue":"0"
            },
            "settable":true
         },
         "eventDetectors":[
         ],
         "plotType":"SPLINE",
         "unit":"",
         "templateXid":"Numeric_Default",
         "chartColour":"black",
         "chartRenderer":{
            "type":"IMAGE",
            "timePeriodType":"DAYS",
            "numberOfPeriods":1
         },
         "dataSourceXid":"DS_362292",
         "defaultCacheSize":1,
         "deviceName":"bugtest",
         "discardExtremeValues":false,
         "discardHighLimit":1.7976931348623157E308,
         "discardLowLimit":-1.7976931348623157E308,
         "intervalLoggingPeriod":1,
         "intervalLoggingSampleWindowSize":0,
         "overrideIntervalLoggingSamples":false,
         "purgeOverride":false,
         "purgePeriod":1,
         "readPermission":"",
         "setPermission":"",
         "textRenderer":{
            "type":"ANALOG",
            "useUnitAsSuffix":true,
            "unit":"",
            "renderedUnit":"",
            "format":"0.00"
         },
         "tolerance":0.0
      }
   ],
   "pointLinks":[
      {
         "xid":"PL_247557",
         "sourcePointId":"DP_829661",
         "targetPointId":"DP_171102",
         "event":"CHANGE",
         "logLevel":"NONE",
         "disabled":false,
         "script":"",
         "scriptPermissions":{
            "customPermissions":"",
            "dataPointReadPermissions":"superadmin",
            "dataPointSetPermissions":"superadmin",
            "dataSourcePermissions":"superadmin"
         },
         "writeAnnotation":false
      },
      {
         "xid":"PL_061881",
         "sourcePointId":"DP_829661",
         "targetPointId":"DP_432453",
         "event":"CHANGE",
         "logLevel":"NONE",
         "disabled":false,
         "script":"return source.value \/ 2;",
         "scriptPermissions":{
            "customPermissions":"",
            "dataPointReadPermissions":"superadmin",
            "dataPointSetPermissions":"superadmin",
            "dataSourcePermissions":"superadmin"
         },
         "writeAnnotation":false
      }
   ],
   "watchLists":[
      {
         "user":"admin",
         "dataPoints":[
            "DP_829661",
            "DP_171102",
            "DP_432453"
         ],
         "editPermission":null,
         "name":"bugtest",
         "readPermission":null,
         "xid":"WL_042975"
      }
   ]
}
```

terrypacker, 2015-10-20:
Thanks for the complete testbed that saves a lot of time.  Unfortunately on my system there is no problem, Setting a=5 gives me b=5 and c=2.5.  I ran a few tests, first from a completely clean Mango installation then by importing your configuration on a running system.

What OS and Java version are you using?  I have seem some problems with java 7 vs java 8 when they changed the underlying script engine.

jorgempy, 2015-10-21:
ok @terrypacker you were right, in java8 it works, no changes.
OS:  Ubuntu 15.04 x64 
First test (didn't work): openjdk java7
Second test (works): openjdk java8

closed 2015-11-10 by terrypacker