MangoAutomation/ma-core-public #440: Add Logging output to log file for Mango Javascript Data Sources

terrypacker, 2015-10-21:
Connect the script log output into the Log4j output during runtime.  Keep the UI test logging output the same to display onto the page.

Puckfist, 2016-06-07:
Has some problems...

1) Did not see output into main Mango log, which is how the issue reads to me. Was logging at level fatal, data source logging above warn. It certainly logged into that file at that level, but I think part of the motivation is enabling loopy-ness like pointing a log4j data source at a scripting data source? Slash scripts can have large effects so logging everything in one place sequentially makes sense?
2) The log file is overwritten with each stop/start of the data source, meaning the first and last messages in the logs/script-id.log  We probably want to append, no?

Puckfist, 2018-03-09:
Maybe I have misunderstood the intention of the issue to add these log messages into the main log file, but if one wishes to do this then one need add the following to ScriptLog (and initialize the LOG variable)

switch(level) {
            case LogLevel.TRACE :
                LOG.trace(message);
                break;
            case LogLevel.DEBUG :
                LOG.debug(message);
                break;
            case LogLevel.INFO :
                LOG.info(message);
                break;
            case LogLevel.WARN :
                LOG.warn(message);
                break;
            case LogLevel.ERROR :
                LOG.error(message);
                break;
            case LogLevel.FATAL :
                LOG.fatal(message);
                break;
        }

Puckfist, 2018-03-14:
Closing as not going to output the script's log output to the main log. The normal script log exists to keep the main log clean.

closed 2018-03-14 by Puckfist