MangoAutomation/ma-core-public #442: Implement "Let's Encrypt" a free SSL certificate authority

stevededrick, 2015-10-23:
There is a new free SSL authority at https://letsencrypt.org/

I think Mango should support it and use SSL by default

jazdw, 2016-12-02:
@stevededrick 
We were just talking about implementing this in Mango v3. Its worth noting you can use it already, you have to run their client and tell it to use Mango's web directory as the web root. You then have to convert the certificate and key file to a Java key store and restart mango.

Commands to convert to PKCS12 and then JKS
```
openssl pkcs12 -export -in "$certdir/fullchain.pem" -inkey "$certdir/privkey.pem" -out /opt/mango/keystore.p12 -name mango -passout pass:mango
keytool -importkeystore -deststorepass mango -destkeystore /opt/mango/keystore.jks -srckeystore /opt/mango/keystore.p12 -srcstoretype PKCS12 -alias mango -srcstorepass mango
rm /opt/mango/keystore.p12
chmod 600 /opt/mango/keystore.jks
```

env.properties
```
ssl.on=true
ssl.port=8443
ssl.keystore.location=/opt/mango/keystore.jks
ssl.keystore.password=mango
```

Puckfist, 2018-04-05:
To do this properly, I think one can only self-sign over HTTP, then can submit a certificate chain and key to a Mango iff it's over HTTPS? Should we try to assess local connections? Should there be a checkbox that says "I don't care if this transmission of security information isn't secure"? While it does seem somewhat unlikely the HTTP submission of the private key would get snorted, isn't that still too risky?