MangoAutomation/ma-core-public #443: Meta Data Point History Generator - No Feedback

terrypacker, 2015-10-23:
no good feedback from generate history window for Meta Data Points

resotek, 2015-10-30:
I overlooked the start year on a "Generate History" entry, and the job created over 1 million and counting medium priority work items. My important metatdata points stopped updating, and there was no way to terminate the job, let alone assess its completion. I had to kill mango, and now I have to wait for it to clean the TSDB. Can you please display the time stamp every 1 or 5 seconds,  and more importantly, support a cancel button?

resotek, 2015-10-30:
After the Meta Data Point History Generator completes a job, it displays in a pop-up window the number of points generated. However, that number is much greater than the number of points generated and added to the pointvalue log. Can the completion dialog list the number of points added to the log, and wording be made clearer regarding the meaning of the large number currently displayed in that dialog?

terrypacker, 2015-12-03:
This issue should be dealt with at the same time:  #318

terrypacker, 2015-12-03:
Also deal with #319

Puckfist, 2018-01-31:
This and #318 are somewhat addressed in the rest controller I put into the Meta-history-generation-API branch of the private modules. It supports getting status by seeing the last processed update's timestamp and the number of points in the submitted request. It supports cancelling the task between generating points' histories

Puckfist, 2018-05-07:
The history generation API is released in the latest meta module

closed 2018-05-07 by Puckfist