MangoAutomation/ma-core-public #446: Persistent TCP event emails do not provide datasource name

resotek, 2015-11-03:
The Persistent TCP Publisher emails typically report something like:
   Event active notification ** Urgent ** 01:38:37 - Connection timed out
or
  Event active notification ** Urgent ** 01:37:34 - Write timeout

This is problematic because from the email we don't know which module is generating the event email. In fact, it's not even obvious that the event was from a Persistent TCP Publisher. When I first received these emails, I thought they were from a datasource.

FIX: Please include the module name and the publisher name in the event emails.

terrypacker, 2015-12-07:
Added in 2.7.0 release.  These events are of type: PublisherEventType so they should be discoverable in that way but there is currently no way to filter on Event Type.  I'm leaving this open and adding to 2.7.x bugs so that we can re-vist the Alarms Page and see if we can improve event filtering for this.

Puckfist, 2015-12-22:
The publisher name is definitely in the event text now.

closed 2015-12-22 by Puckfist