MangoAutomation/ma-core-public #447: Z-Wave Data Source Error Handling

terrypacker, 2015-11-12:
During startup the Z-Wave Control Center can fail to start and cause problems if it cannot access the device library or load it JNI libs.  There is no fail safe code to prevent this killing the RuntimeManager at startup.

We need to add logging and some event that is generated if this happens instead of just failing on NPE.

terrypacker, 2015-12-22:
To test create a Zwave data source and set to enabled.  Then modify the Zwave protocol library jar to not have native binaries for your system.  Then restart and see errors logged but startup still happens without the datasource running.

Puckfist, 2015-12-22:
Not resolved. A Zwave data source running and pointed at a serial port connection crashes Mango fast from wherever it is

Puckfist, 2015-12-23:
That last post is not true, of course, if that port actually contains a zwave device

Puckfist, 2015-12-23:
When I cleared out the native libs, I only got a RuntimeException at boot, NoClassDefFoundError at poll.

[zwave stack traces.txt](https://github.com/infiniteautomation/ma-core-public/files/70371/zwave.stack.traces.txt)

terrypacker, 2015-12-23:
I'm leaving this open and adding a note that the recent changes may have introduced a bug.  I'm not uploading a new module to the store for 2.7.2 yet.

Stacktrace:
```
ERROR 2015-12-22 12:45:41,212 (com.infiniteautomation.zwave.rt.ZwaveDataSourceRT.onNotification:457) - java.io.StreamCorruptedException: invalid type code: 00 
com.serotonin.ShouldNeverHappenException: java.io.StreamCorruptedException: invalid type code: 00
 at com.serotonin.util.SerializationHelper.readObjectInContext(SerializationHelper.java:93)
 at com.serotonin.m2m2.db.dao.DataSourceDao$DataSourceRowMapper.mapRow(DataSourceDao.java:111)
 at com.serotonin.m2m2.db.dao.DataSourceDao$DataSourceRowMapper.mapRow(DataSourceDao.java:108)
 at org.springframework.jdbc.core.RowMapperResultSetExtractor.extractData(RowMapperResultSetExtractor.java:93)
 at org.springframework.jdbc.core.RowMapperResultSetExtractor.extractData(RowMapperResultSetExtractor.java:60)
 at org.springframework.jdbc.core.JdbcTemplate$1.doInPreparedStatement(JdbcTemplate.java:693)
 at org.springframework.jdbc.core.JdbcTemplate.execute(JdbcTemplate.java:629)
 at org.springframework.jdbc.core.JdbcTemplate.query(JdbcTemplate.java:680)
 at org.springframework.jdbc.core.JdbcTemplate.query(JdbcTemplate.java:712)
 at org.springframework.jdbc.core.JdbcTemplate.query(JdbcTemplate.java:722)
 at com.serotonin.db.spring.ExtendedJdbcTemplate.queryForObject(ExtendedJdbcTemplate.java:39)
 at com.serotonin.db.DaoUtils.queryForObject(DaoUtils.java:332)
 at com.serotonin.m2m2.db.dao.AbstractBasicDao.get(AbstractBasicDao.java:317)
 at com.serotonin.m2m2.vo.permission.Permissions.hasEventTypePermission(Permissions.java:253)
 at com.serotonin.m2m2.rt.EventManager.raiseEvent(EventManager.java:119)
 at com.serotonin.m2m2.rt.dataSource.DataSourceRT.raiseEvent(DataSourceRT.java:150)
 at com.infiniteautomation.zwave.rt.ZwaveDataSourceRT.onNotification(ZwaveDataSourceRT.java:174)
 at com.infiniteautomation.zwave.rt.MangoZwaveDriver.onNotification(MangoZwaveDriver.java:99)
 at com.infiniteautomation.zwave.rt.ZwaveControlCenter.onNotification(ZwaveControlCenter.java:93)
Caused by: java.io.StreamCorruptedException: invalid type code: 00
 at java.io.ObjectInputStream.readObject0(ObjectInputStream.java:1379)
 at java.io.ObjectInputStream.defaultReadFields(ObjectInputStream.java:1993)
 at java.io.ObjectInputStream.readSerialData(ObjectInputStream.java:1918)
 at java.io.ObjectInputStream.readOrdinaryObject(ObjectInputStream.java:1801)
 at java.io.ObjectInputStream.readObject0(ObjectInputStream.java:1351)
 at java.io.ObjectInputStream.readObject(ObjectInputStream.java:371)
 at com.serotonin.util.SerializationHelper.readObjectInContext(SerializationHelper.java:90)
 ... 18 more
```