MangoAutomation/ma-core-public #449: Clarify what the upgrade backs up

resotek, 2015-11-13:
After loading the modules page and clicking on 'Check for upgrades' button, there is a backup option:
* Advanced options (hide)
* Do backup before download

What exactly does "do backup" back up? 
Just the modules? 
The configuration database?
The noSQL database?

Can we clarify what is backed up, and to where, and what is not backed up?
e.g. "Backup modules to <mango_dir>/backup before download"
or "Backup modules and SQL configuration to <mango_dir>/backup before download"
or "Backup modules and SQL configuration and SQL PointData to <mango_dir>/backup before download. NoSQL PointData is not backed up."

Joelhaggar, 2015-11-13:
It's going to run the standard backups you have configured on the system setting page.  If using H2 and NoSQL then those two database and the Configuration Export will all run.  If you are using MySQL it will not be backed up.  Note, at this time none of the file system will be backed up.

terrypacker, 2015-12-03:
Add a help icon into the Upgrade area to explain the options/settings.

closed 2016-01-31 by Puckfist