MangoAutomation/ma-core-public #454: Excel Reports Device Name Search not performing correct

Joelhaggar, 2015-11-27:
See forum post: http://infiniteautomation.com/forum/topic/2029/excel-reports-add-a-point-functionality

[image]

[image]

Puckfist, 2015-12-07:
This is still an issue.

1. Filter by.... is editable. This means I can erase 'me' in Name and get silent server errors.

Puckfist, 2015-12-13:
[excel reports stack trace.txt](https://github.com/infiniteautomation/ma-core-public/files/60348/excel.reports.stack.trace.txt)

Puckfist, 2015-12-19:
Closed. It appears the search has changes to be name, device name, or XID, but only one at a time.

closed 2015-12-19 by Puckfist