MangoAutomation/ma-core-public #455: Limit view of Users to Email in Reports

terrypacker, 2015-11-30:
Use permissions to limit the view of users when adding users to the email list of a report.

terrypacker, 2015-12-01:
Change so that Users can see other users if they have at least 1 group in common, admin users can still see all users.  Keep Mailing lists viewable by all users.

Puckfist, 2015-12-07:
This fix is working in the sense that only users who can be seen by the creator of the report are those who share a group, as well as all the mailing lists. Some bad behavior:

If a report has more users that are supposed to be emailed than a particular person is permitted to view, they will load the report page with only the users they are permitted to view. If they save that, the email recipients is now only the list that person saw.

terrypacker, 2016-05-26:
See issue #640 for a general solution to the missing permissions problem.

terrypacker, 2016-06-02:
As #640, don't remove users that the saving user doesn't have permissions to see.

Potentially provide message that users exist and you can't see them.