MangoAutomation/ma-core-public #456: ES Config Module IPv4 Subnet

Puckfist, 2015-11-30:
I believe the subnet is getting set twice during inbound conversion to update interface settings. The first call correctly decodes the string mask and sets networkPrefixLength in an InterfaceAddressVO, but a second call is made during inbound conversion to set the networkPrefixLength directly, and this value is the lingering value from the outbound conversion of the interface. To fix, set the networkPrefixLength to -1 explicitly (such that it performs a no-op, InterfaceAddressVO:30) at systemManager:349-355,     iface.ipv4Address.networkPrefixLength = -1;

terrypacker, 2015-12-17:
The above fix worked, thanks.  I've found another problem where the Subnet mask isn't being re-set correctly, although this may just be a one off side effect of my tests.

To reproduce:

1. Set Subnet mask to 255.255.255.255 via the ES config page
2. Change /etc/network/interfaces back with: net mask 255.255.255.0
3. sudo ifdown eth0
4. sudo ifup eth0
5. ifconfig and see that the net mask is 255.255.255.255
6. sudo ifdown and sudo ifup and see the net mask at 255.255.255.0

terrypacker, 2015-12-18:
Should see better results now as per this commit: 56dcd266d053f0d1d933f4a9ac0c239a9b98836f

Added some validation too.

Puckfist, 2015-12-18:
Looks good.

closed 2015-12-18 by Puckfist