MangoAutomation/ma-core-public #458: Thread Pool Size instabilities

Puckfist, 2015-12-01:
Setting everything in the thread pool settings to 1 will prevent Mango from shutting down or booting up. This setting is stored in the database. Setting the max high priority threads to 1 never seems like a good idea, and even 2 is harrowing if the system is doing anything.

Through the import/export utility, no validation is performed for the pool size. Negative numbers, zero, and numbers which overflow a signed 32 bit integer are all accepted. Setting medium pool size and max medium or low size and max low to renders Mango incapable of starting. Setting it to the overflow 1<<32 seems to take into the database, but doesn't seem to apply to the systems settings page or in the program (still stable). I didn't see an error every go out the console either.

Options for the Max Medium Priority Size and Max Low Priority Size are available to be modified in the JSON export but are not available from the system settings page. System settings appears to set both the max and the size to the value entered.

terrypacker, 2015-12-02:
Ready for test 500f47c8d0ad9eae110e2ba63da776d33d117285

Puckfist, 2015-12-03:
Looks good. One blemish, though, the error messages for the HighPriorityCorePoolSize and the high priority max pool size are stepping on each other. I think the check on the max wants to be displayed below its input.

terrypacker, 2015-12-03:
Nice catch.  That's fixed now too.

Puckfist, 2015-12-07:
Confirmed

closed 2015-12-07 by Puckfist