MangoAutomation/ma-core-public #460: WebSocket Interruption causes ballooning medium work items

Puckfist, 2015-12-02:
If a page is using websockets and an interrupt occurs in the client's connection (i.e. a hardware switch on the wireless) the sessions do not close properly on the server side and medium priority publishing tasks climb until system badness.

To see, use a configuration like the attached, then open the pointValuesWebSocket tutorial for custom dashboards. Register all the points, and through the hardware switch for your connection or unplug the cable. Close the browser tab that had the pointValuesWebSocket tutorial open. If you don't, it may reconnect and drain the medium priority pool. Now you can reestablish your internet connection, or use a second computer to log into the mango. You're likely to see all medium-pool-2 threads blocked by one of their own.

[12-2-15 test.txt](https://github.com/infiniteautomation/ma-core-public/files/50155/12-2-15.test.txt)

terrypacker, 2015-12-02:
I've found on local testing that after about 10 minutes of breaking the connection the recently added code from this commit e61cc0e42da94f7ade489b0efa63e39fec5fda6d  kicks in and closes the sockets.  I will do some additional investigation to see why 10 minutes is the amount of time.

terrypacker, 2015-12-02:
We will need to implement a Ping-Pong message from Mango to the Browser to do any better until we move to Jetty 9.3.x where they have implemented timeouts on writes via Futures (Java 8 Only).

terrypacker, 2015-12-04:
After further investigation, Futures can be used on Java 7 but they did not fix the problem.  The future seems to say the message was sent before the method returns and un-locks the sync block.  Ping/Pong connection testing seems like the way to go, I'm implementing that now to test.

terrypacker, 2015-12-04:
Ping/Pong tracking seems to work nicely.  Adding optional PingPong tracking for all sessions, its used by default and the timeout is set to 10s.  Not sure if it should be run as a High priority task but it is for now.

terrypacker, 2015-12-04:
When testing this keep in mind:

1.  The pong response is handled by the web browser (no code we have written).  So testing different browser types will be important. 

2.  Not sure how proxies deal with these messages, if they are discarded the web socket connections will be terminated.

Default is 10s timeout but it's adjustable via the env.properties file for now.

terrypacker, 2015-12-04:
Also note this commit infiniteautomation/ma-modules-private@e3933f7db6049a381b4d5a35390ab9c67d0e385d

Puckfist, 2015-12-07:
Not resolved. The issue still seems to occur after an upgrade. The reason could be related to my already having a MA_HOME/overrides/properties/env.properties file which does not contain the timeout settings. It seemed to work fine when everything was in the env.properties file as expected, but we cannot count on them being into the expected env.properties

terrypacker, 2015-12-08:
Another fix using Jetty asynchronous writes to the socket to avoid deadlock when the write hangs on a dead connection.
See commit:
39f34839fae5ca527e06edba3d7d790d48990684

Puckfist, 2015-12-08:
Resolved

closed 2015-12-08 by Puckfist