MangoAutomation/ma-core-public #461: Users Page not displaying Server Time Zone

Puckfist, 2015-12-02:
When set to 'use server time zone' it does not display that time zone's value beneath the select.

terrypacker, 2015-12-02:
If I'm understanding this correctly I think this is supposed to work that way.

The 'use server time zone' is its own unique setting.  For example if all users are set to this and the server moves timezones then all the users move timezones too.  It doesn't auto-detect the timezone and set it to a specific zone.

Puckfist, 2015-12-02:
Ah, perhaps I misunderstood what "Time zone label" meant in line 7 of our spreadsheet, for the issue at https://github.com/infiniteautomation/ma-core-public/commits/91953a502717dc901046ba70e69b9c9701f052b9

I was interpreting that to mean it would say Zone/City under the timezone, not that I think it should or shouldn't. Perhaps it just meant the label for the table section

terrypacker, 2015-12-03:
yes I just wanted verification that when using the 'use server time zone' option everything still works.

closed 2015-12-08 by Puckfist