MangoAutomation/ma-core-public #464: CSV Import "pointLocatorType" Not Important

Puckfist, 2015-12-02:
Using the configuration attached, you can do a CSV import of

"modelType","xid","name","enabled","deviceName","dataSourceXid","readPermission","setPermission","templateXid","unit","useIntegralUnit","integralUnit","useRenderedUnit","renderedUnit","pointLocatorType","dataType","settable","relinquishable","changeType","maxChange","volatility","attractionPointXid","min","max","change","roll","values","amplitude","offset","period","phaseShift","startValue"
"DATA_POINT","FastOrigin_Brownian_1","Brownian_1","True","FastOrigin","FastOrigin","","","Numeric_Default","","False","s","False","","PL.VIRTUAL","NUMERIC","False","False","BROWNIAN","10.0","0.0","","0.0","100.0","0.0","False","[ ]","0.0","0.0","0.0","0.0","50"
"DATA_POINT","FastOrigin_Brownian_1","Brownian_1","True","FastOrigin","FastOrigin","","","Numeric_Default","","False","s","False","","APL.AVIRTUAL","NUMERIC","False","False","BROWNIAN","10.0","0.0","","0.0","100.0","0.0","False","[ ]","0.0","0.0","0.0","0.0","50"

And both will save. The second's pointLocatorType, APL.AVIRTUAL, has some extra A's in it. I can't have such a value in the first line, but the second accepts it.

Puckfist, 2015-12-02:
I have discovered that import DOES fail when the headers have been removed, and says APL.AVIRTUAL is not well formed.

terrypacker, 2015-12-02:
This is due to the fact that when importing data CSV it all must be of the same type, so currently the first Point Locator Type is used to construct a handler that imports all the remaining rows.  The handler is constructed prior to importing the first line of data. The code knows it needs a special handler because the data type for that column has a class that is annotated with CsvEntity(derived=true).

The CSV rest message handling should be refactored and cleaned up at some point.

terrypacker, 2015-12-03:
Fixing the missing headers problem by checking that the first column of the first row is 'modelType'.  This is ok for now because we only use CSV models based on AbstractVoModel with this converter.

terrypacker, 2015-12-03:
Fixes here: 039ec5b78700c454ead7cdf6a16615a19e3bcfa3

closed 2015-12-10 by Puckfist