MangoAutomation/ma-core-public #465: CSV Import Header length > row length

Puckfist, 2015-12-02:
An ArrayIndexOutOfBoundsException is thrown when there are more headers than columns in the data. It manifests as a 500 error printed stack trace to the page, but the ArrayIndexOutOfBoundsException line only displays the index at fault, which is not informative. To evince, do a CSV export of any data points, add a comma to the end of the headers and hit import.

terrypacker, 2015-12-03:
Cleaned up the response messages a little and added checks for every row agains the length of the header.  Since the header is assumed to define the document the best we can do is let the user know that a row is different in length.  There is an error for row too long and one for row too short.

Puckfist, 2015-12-08:
It still prints a stack trace, but it is more informative

closed 2015-12-10 by Puckfist