MangoAutomation/ma-core-public #466: CSV Import - Row Error order matters?

Puckfist, 2015-12-02:
Certain errors can be handled in the first data point row, while other errors seem to bring up a 500 error and a stack trace. Oddly enough, those same data point forming errors can be done in subsequent rows. Using the configuration supplied,

This import will produce the error: 
"modelType","xid","name","enabled","deviceName","dataSourceXid","readPermission","setPermission","templateXid","unit","useIntegralUnit","integralUnit","useRenderedUnit","renderedUnit","pointLocatorType","dataType","settable","relinquishable","changeType","maxChange","volatility","attractionPointXid","min","max","change","roll","values","amplitude","offset","period","phaseShift","startValue"
"DATA_POINT","FastOrigin_Brownian_2","Brownian_2","True","FastOrigin","FastOrigin2","","","Numeric_Default","","False","s","False","","PL.VIRTUAL","NUMERIC","False","False","BROWNIAN","10.0","0.0","","0.0","100.0","0.0","False","[ ]","0.0","0.0","0.0","0.0","50"
"DATA_POINT","FastOrigin_Brownian_1","Brownian_1","True","FastOrigin","FastOrigin","","","Numeric_Default","","False","s","False","","PL.VIRTUAL","NUMERIC","False","False","BROWNIAN","10.0","0.0","","0.0","100.0","0.0","False","[ ]","0.0","0.0","0.0","0.0","50"

Notice the dataSourceXid in the first point is not valid. Switch the rows and you'll get a pretty output of that fact.

[12-2-15 test.txt](https://github.com/infiniteautomation/ma-core-public/files/50256/12-2-15.test.txt)

terrypacker, 2015-12-03:
Fixed by ensuring we have a reference data source (usually the first row's dataSourceXid) before we can start inserting points.  If the first data source is missing then it will show up as invalidXid.  Once a good data source xid is found all following points are compared to it to ensure they are of that type (which is a requirement of this endpoint for now).

Puckfist, 2015-12-08:
True, I guess to exhaustively test would be to error every field and switch them. Perhaps i could figure out what are the special positions and fields with a little reading, and direct some testing at those.

Confirmed that the producer of my overly generalized issue is now handled.

closed 2015-12-10 by Puckfist