MangoAutomation/ma-core-public #467: CSV Import - Gets Stuck, no feedback

Puckfist, 2015-12-02:
Using the attached configuration and import, you can produce an unending, unerring "Import in progress..."

"modelType","xid","name","enabled","deviceName","dataSourceXid","readPermission","setPermission","templateXid","unit","useIntegralUnit","integralUnit","useRenderedUnit","renderedUnit","pointLocatorType","dataType","settable","relinquishable","changeType","maxChange","volatility","attractionPointXid","min","max","change","roll","values","amplitude","offset","period","phaseShift","startValue"
"DATA_POINT","FastOrigin_Brownian_2","Brownian_2","True","FastOrigin","FastOrigin","","","Numeric_Default","","False","s","False","","PL.VIRTUAL","NUMERIC","False","False","BROWNIANS","10.0","0.0","","0.0","100.0","0.0","False","[ ]","0.0","0.0","0.0","0.0","50"
"DATA_POINT","FastOrigin_Brownian_1","Brownian_1","True","FastOrigin","FastOrigin","","","Numeric_Default","","False","s","False","","PL.VIRTUAL","NUMERIC","False","False","BROWNIAN","10.0","0.0","","0.0","100.0","0.0","False","[ ]","0.0","0.0","0.0","0.0","50"

Notice the change type has been set to BROWNIANS, which does not exist.

I have produce the same issue, albeit in a different way, but importing:
"modelType","xid","name","enabled","deviceName","dataSourceXid","readPermission","setPermission","templateXid","unit","useIntegralUnit","integralUnit","useRenderedUnit","renderedUnit","pointLocatorType","dataType","settable","relinquishable","changeType","maxChange","volatility","attractionPointXid","min","max","change","roll","values","amplitude","offset","period","phaseShift","startValue"
"DATA_POINT","FastOrigin_Brownian_2","Brownian_2","True","FastOrigin","FastOrigin","","","Numeric_Default","","False","s","False","","PL.VIRTUAL","NUMERIC","False","False","BROWNIAN","10.0","0.0","","0.0","100.0","0.0","False","[ ]","0.0","0.0","0.0","0.0","50"
"DATA_POINT","SlowOrigin_Comp_Average_1","Average_1","True","SlowOrigin_Computer","SlowOrgin_Computer","","","Numeric_Default","","False","s","False","","PL.META","NUMERIC","False","False","return p1.past(MONTH).average;","JAVASCRIPT","CONTEXT_UPDATE","","0","my","[ {
  ""dataPointXid"" : ""SlowOrigin_Brownian_1"",
  ""variableName"" : ""p1"",
  ""contextUpdate"" : true
} ]","{
  ""dataSourcePermissions"" : ""superadmin"",
  ""dataPointSetPermissions"" : ""superadmin"",
  ""dataPointReadPermissions"" : ""superadmin"",
  ""customPermissions"" : """"
}"

Note here that I've naively copied a row for a meta point beneath a row for virtual, beneath the headers for virtual.

[12-2-15 test.txt](https://github.com/infiniteautomation/ma-core-public/files/50256/12-2-15.test.txt)

terrypacker, 2015-12-03:
Cleaned up error detection for various cases when setting a value fails also cleaned up Virtual Point Locator model to provide better feedback and not throw ShouldNeverHappenExceptions when something goes wrong.  

Other models should be reviewed for this infiniteautomation/ma-modules-public@ 7f7490be39b5c328e07dadba3e50d0515ad4b3d1 but even if they fail the new core code should provide much better feedback than before.

Puckfist, 2015-12-10:
Moving to 2.7.x milestone until more model's point locators' property sets are tested. Confirmed it catches the two causes I had in this issue.

closed 2015-12-21 by Puckfist