MangoAutomation/ma-core-public #468: CSV Import - Cast Exception ugly output

Puckfist, 2015-12-02:
If a field is supplied with a value which cannot be cast to its sought type, the emport page sends code 500 and a stack trace, yet missing the name of the exception. To evince, import any CSV for a data point with a modification that would render a field unable to be parsed.

i.e.
"False" --> "Falser"
"0.0" --> "Cheeseburger"

terrypacker, 2015-12-03:
Further cleaned up error handling for CSV import.  The main issue is that since these types of errors are during the model conversion we never get a model to validate so we cannot get back a list of validation errors to display in the list.

Puckfist, 2015-12-08:
These errors still output a stackstrace to the page.

Puckfist, 2015-12-10:
Closed, looks clean.

closed 2015-12-10 by Puckfist