MangoAutomation/ma-core-public #469: New Data Point Details - Still refreshing after logout

Puckfist, 2015-12-02:
Using two tabs, login and view a point that updates with some regularity on the new Data Point Details view. Bring up any page in your other tab, and use it to log out. Values are still updating!

Puckfist, 2015-12-02:
The same is true for other websocket powered pages as well, such as the pointValuesWebSocket

terrypacker, 2015-12-03:
One option is SpringSession: https://github.com/spring-projects/spring-session/tree/1.0.2.RELEASE//samples/security

Still investigating why the web socket session isn't getting closed on logout though.

terrypacker, 2015-12-03:
Added code to close WebSocket after HttpSession is invalidated, made this the default action but provided the option to create WebSockets that remain open after logout.

Puckfist, 2015-12-08:
Confrimed

closed 2015-12-08 by Puckfist