MangoAutomation/ma-core-public #470: Static html/shtm files not rendering with Content-Type set.

terrypacker, 2015-12-03:
Not exactly sure why but the Spring interface org.springframework.web.servlet.View is not adding the Content-Type header into the response for the Custom Dashboards module.  This has been fixed in this commit infiniteautomation/ma-modules-private@6ff3dccd59f908e20dae4cc1f8ba5833f097d241

But we need to verify any other problems like this that might arise, a quick review of where we use the View interface (although not all apply to this problem):

* Custom Dashboards
* DGLux
* Graphical Views
* JSP Views
* Reports
* SQL Console
* Watchlist
* Core

terrypacker, 2015-12-07:
Fixed Reports here: infiniteautomation/ma-modules-public@5dba2dbf05d432a252cb32e663577c14ec284b04

Puckfist, 2015-12-08:
Custom Dashboards, Graphical Views, Reports, SQL Console, Watchlist and the core seem to be behaving. and i'm about to make a new issue for JSP Views...

I haven't tested DGLux (since it is flash and doing its own stuff in the DWR for its module, it seems fairly safe to assume this shouldn't affect it. I have confirmed the .swf still loads).

Puckfist, 2015-12-09:
The JSP Views item, [this issue](https://github.com/infiniteautomation/ma-core-public/issues/497), was unrelated and resolved.

Puckfist, 2015-12-21:
Everything is seemingly working

closed 2015-12-21 by Puckfist