MangoAutomation/ma-core-public #471: Permissions can be given to whitespace

Puckfist, 2015-12-03:
Not really a bug per se, but you can have a user group composed of only spaces. Since we're trimming, all user groups composed of only spaces are equivalent.

terrypacker, 2015-12-03:
Added validation to check for empty permissions when saving a User.  

This will catch things like ``` , , ``` but not like ``` ,, ```

Puckfist, 2015-12-03:
It seems like ,, doesn't have the issue since emptiness is discarded before the trim operation

Puckfist, 2015-12-08:
Confirmed cannot save whitespace as a user's permission group, but whitespace can still be a group in other permissions fields.

Puckfist, 2015-12-09:
Marking as closed since the issue cannot be evoked anymore, but there are still opportunities to clean up validation on other permissions fields.

closed 2015-12-09 by Puckfist