MangoAutomation/ma-core-public #476: New Data Point Details - Useless refresh button

Puckfist, 2015-12-03:
It's pretty, but it seems disconnected. With the way we're updating the page now, it doesn't have a purpose, does it?

terrypacker, 2015-12-03:
That button refreshes the data point by calling DataSourceRT.forcePointRead().  This is not required to do anything for all data sources but some have this implemented.  So this button doesn't actually refresh the view it refreshing the backing data.  If implemented properly you could essentially request point values from a  data source with having it poll.

terrypacker, 2015-12-03:
I'll double check that its actually wired up, it may not be as I can't remember so if not I will get it working if possible.

terrypacker, 2015-12-09:
This should be ready to test now.

Puckfist, 2015-12-09:
Confirmed the functionality is working, but the img is not in button mode (clickable but not giving me the button styled point, merely a standard cursor as I hover over it). Perhaps it needs the 'clickable' class?

Puckfist, 2015-12-19:
Clearly clickable

closed 2015-12-19 by Puckfist