MangoAutomation/ma-core-public #483: Query Logs Help Documents not extremely helpful

Puckfist, 2015-12-05:
Here is the text:
========
RQL Queries for Logfiles

This terminal allows performing queries on the ma.log file. The query syntax is RQL. Always use a reasonable limit to avoid memory issues in the browser or Mango.

Query Examples

level=INFO&limit(10)
(class=INFO|class=DEBUG)&limit(10)
like(classname,.*m2m2.*)&limit(10)
========

Complaints:
1) Headers are presented with a capital letter, but using a capital letter to filter the field doesn't work.
2) The second example is odd. INFO and DEBUG (which do not need case sensitivity) strike me as level= properties. This is further compounded by the third item...
3) classname is not a column in the table. Class is, but filtering by "class=com.serotonin.m2m2.meta.MetaPointLocatorRT&limit(10)" does nothing, it needs to be classname.

Other observations,
1) Cannot use line=# or line># (whether or not you would want to...)
2) limit(10) has to be the last element? Doing limit(10)&line=180 will fill my whole table, but line=180&limit(10) will only fetch 10.
3) What would be the search header for stack trace? Is there one?

terrypacker, 2015-12-07:
Complaints have been fixed by changing the help text to actually make sense.

Observations:
1) RQL for line number was not implemented.
2) RQL enforces that rule.
3) RQL for Stack Trace was not implemented.

terrypacker, 2015-12-07:
Reference to commit: infiniteautomation/ma-modules-private@934593fa6e3a5840cc7f92687356197a5c12bb52

Puckfist, 2015-12-08:
Confirmed, helpful and observant

closed 2015-12-08 by Puckfist