MangoAutomation/ma-core-public #484: Email logs to IAS

Puckfist, 2015-12-05:
"Log File sent to Infinite Automation" is immediately given back to the user as fact, and the only whisper that my SMTP is not configured is an information event easily overlooked. This is similar to other test email buttons, which claim to send whether or not SMTP is configured.

terrypacker, 2015-12-06:
This is because the email is sent as a work Item that is processed in another thread and not directly when the button is pressed.  The fix is to implement the 'after send email' method in the Email Work Item for all these buttons so they can provide feedback to the page.  This will also require polling by the page to pickup the results.

Puckfist, 2018-01-30:
Closing as unimportant

closed 2018-01-30 by Puckfist