MangoAutomation/ma-core-public #485: Excel Reports - Unclear dependency message

Puckfist, 2015-12-06:
Without mango's API module, navigating to the Excel Reports page yields large black letters proclaiming:

Required module '{0}' does not exist

terrypacker, 2015-12-09:
This is a 2 part fix that we will do after the 2.7.0 release.

Part 1 in core, change
``` common.moduleDoesNotExist=Required module '{0}' does not exist ```
to
``` common.moduleDoesNotExist=Required module {0} does not exist ```

Part 2 in module, change line 295 in excelReports.jsp from
``` <fmt:message key="common.moduleDoesNotExist"> ```
to
```
 <h1 style="color:red">
    <fmt:message key="common.moduleDoesNotExist">
      <fmt:param value="mangoApi"/>
    </fmt:message>
  </h1>

```

Puckfist, 2015-12-09:
resolved

closed 2015-12-09 by Puckfist