MangoAutomation/ma-core-public #486: Excel Reports - Silent Dependency

Puckfist, 2015-12-06:
Without the custom dashboards module installed, but with the api module, loading the page errors to the console and is otherwise visible only in lack of functionality.

Error was:
http://localhost:8080/mango-javascript/mango-2.0/api.js Failed to load resource: the server responded with a status of 404 (Not Found)
require.js:8 Uncaught Error: Script error for: mango-2.0/api

terrypacker, 2015-12-06:
The module dependency functionality of the module.properties file has a bug that has been fixed in the next version of the store.  Until then we will need to add some jsp code to inform the user that the module is missing.

terrypacker, 2015-12-09:
We should also expand the ModuleExistsTag and ModuleDoesNotExistTag to allow multiple module names separated on a comma.

Puckfist, 2015-12-09:
Resolved

closed 2015-12-09 by Puckfist