MangoAutomation/ma-core-public #487: Excel Reports - Saving without points quiet failure

Puckfist, 2015-12-06:
Saving a report with no points produces a validation error message at the top, but no information as to the error.

terrypacker, 2015-12-09:
The validation comes back with the correct message its just not getting applied.  Here is the message in the JSON response:
```
  "validationMessages" : [ {
    "message" : "At least one point must be added",
    "level" : "ERROR",
    "property" : "timeSeries[0].points"
  } ],
```

The fix might be to set the correct class for each time series in the UI to match timeSeries[x].points.  The current class is set to time-series-points.

terrypacker, 2015-12-09:
We should also fix the BaseUIComponent javascript class to handle the situation when it cannot find the property in the validation message and display some generic message with this info so this is easier to identify as a bug.

terrypacker, 2015-12-17:
You will need to pickup changes from the CustomDashboards, and Core too see all fixes work.

Puckfist, 2015-12-17:
Confirmed

closed 2015-12-17 by Puckfist