MangoAutomation/ma-core-public #491: Excel Reports - Validation says success, report instance says failure

Puckfist, 2015-12-07:
If something goes wrong in the filling out of the excel template, the validation message can still say success even if running the report failed. To evince, use the demo templates in the excel module and only add one point assigned to an invalid field. Run the report and watch the top of the page. You should see a green validation saying "Success!" but if you click to run reports it will have failed.

terrypacker, 2015-12-09:
We will need to discuss this further.  Since reports are run in the background the 'Success' you are seeing is that the report has been scheduled and run in the background.  We won't know it has failed until it actually runs, which is after the REST call to run the report has returned.  One option would be to display a message that says 'Scheduled' or 'Running' instead of Success.

terrypacker, 2016-05-26:
Is this important or should we close it?

terrypacker, 2016-06-02:
Change to 'Scheduled'

Puckfist, 2018-09-21:
The new UI says "Running"

closed 2018-09-21 by Puckfist