MangoAutomation/ma-core-public #494: Graphical Views - User Permissions Inadequate

Puckfist, 2015-12-08:
With the system settings:

Graphical View Permissions: user
Graphical View Upload Permissions:

My test user catfish (groups: user,catfish,brittle,moon) can edit my graphical view, create new graphical views, copy my graphical view, save modifications to my graphical view's permissions section (although only using their user's groups (meaning if I have another permission in there they will see it but saving will be impossible until that user group is removed). What's worse, catfish can do this all even if my graphical view never gave any sort of permission (read, set, or edit) to its groups.

I think we need another System Settings tier for create/copy functions. Untying the knot of whether a lower user should be able to modify the permissions may take more thought, but it doesn't seem right that they need to be a part of every group in the permissions area to save it.

terrypacker, 2015-12-09:
Part I
This is a rule we have enforced, you cannot give permissions to a group you are not in.  I'm open to discussion on this.

Part II

I cannot replicate this.  The permissions seem to be enforced for each level.  We will need to discuss this further.

Puckfist, 2015-12-09:
Resolved, sorry if it was distracting. It had witnesses,

closed 2015-12-09 by Puckfist